Koenkk/zigbee-OTA #644: IKEA E1745 does not update

emilianogetino, 2025-01-07:
Hello, I have an Ikea E1745 zigbee motion detector that does not update the firmware.
Neither from the HA configuration nor from Z2M in the OTA option, see attached images.
[image]
[image]
[image]
[image]
[image]

Is this happening to anyone else? How can I fix it?
All the best.

h1nnak, 2025-01-07:
Look here, this will be the reason.
https://github.com/Koenkk/zigbee2mqtt/issues/25587

ashimokawa, 2025-01-07:
@emilianogetino 

Be sure to walk in front of it to wake it before pressing the button.

emilianogetino, 2025-01-08:
> @emilianogetino
> 
> Be sure to walk in front of it to wake it before pressing the button.

Certainly that was the problem, we had to wake him up.
To wake him up by detecting movement was not enough, I had to pair him again and right at the end of the interview give him the update.
After several attempts it finally started updating and two hours later I already had the new version and it works perfectly.
Thanks for the help.
All the best.

closed 2025-01-08 by emilianogetino