Koenkk/zigbee-OTA #649: Third Reality Humidity Sensor Update Not Found

shall4, 2025-01-12:
Third Reality Humidity Sensor prompted an OTA update from v.30 to .47, but it fails with a 404.

`[2025-01-11 15:57:44] info: 	z2m: Updating 'Laundry Humidity ' to latest firmware
[2025-01-11 15:58:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Laundry Humidity ', payload '{"battery":100,"humidity":49.15,"linkquality":36,"temperature":20.3,"update":{"installed_version":30,"latest_version":47,"state":"available"}}'
[2025-01-11 15:58:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'Laundry Humidity ' failed (Invalid response from https://github.com/Koenkk/zigbee-OTA/raw/master/images/ThirdReality/ThermalLiteSensor_PROD_OTA_V47_1.00.47.ota status=404.)","status":"error"}'
[2025-01-11 15:58:56] error: 	z2m: Update of 'Laundry Humidity ' failed (Invalid response from https://github.com/Koenkk/zigbee-OTA/raw/master/images/ThirdReality/ThermalLiteSensor_PROD_OTA_V47_1.00.47.ota status=404.)`

tyrel, 2025-01-12:
It looks like 47 was merged then reverted (reason unclear) but not completely.

See https://github.com/Koenkk/zigbee-OTA/pull/648#issuecomment-2585289980

I'm curious if there's an intermediate version (46? 45?) that would be good to use if 47 is bad. Rather than going back to 30 (factory) or 29 (this repo).

tyrel, 2025-01-12:
It appears that Third Reality has released version 56 to address the problem with 47.
https://3reality.com/release_note/temperature-and-humidity-sensor-lite/
How one gets that is unclear at this point.
FYI @weihuan1111

Koenkk, 2025-01-12:
Should be fixed now, could you double check?

1st-BrainStormer, 2025-01-12:
Seems to be one version older so can't check update. I Have 5 sensors 1 on Version 44(0x2C) and the other 4 on Version 30(0x1e) Guess we'll have to wait for third reality to make the update available. To bad they don't provide the link on their changelog page...

shall4, 2025-01-12:
I checks as of this morning and a check for update. Installed version is 30 and the latest version is 29. Thanks for looking into it. I guess not much to do here; sensor seems to be working.

1st-BrainStormer, 2025-02-22:
How long is it going to take Third Reality to release the fixed firmware 1.00.56?  I thought they Joined  "Works with Home Assistant"
you'd think they would send the updated firmware when the posted the release notes on January 6

1st-BrainStormer, 2025-09-10:
There have been several core updates.  Have they fixed the OTA issues yet for the Temp/humidity sensor light yet?
@weihuan1111

tyrel, 2026-10-17:
Still waiting for an update here. The Third Reality website says:
>The Home Assistant platform (Zigbee Home Assistant and Zigbee To MQTT) does not yet support OTA firmware updates for this version. Please stay tuned.

We've been staying tuned for over a year and a half.
I'm not sure @weihuan1111 is still involved here, recent PRs on other devices have been from @3reality-support so maybe they can help? For context: Version 47 was released, it had a problem, got reverted. We saw version 56 on the release notes website but it's gone now and 47 is back. Would love to get a new firmware for this device -- the version currently available is older than the version you're shipping with!