Koenkk/zigbee-OTA #676: NodOn Updates fail

tobzsc, 2025-02-06:
Seems like several users have problem updating their NodOn devices. See https://github.com/Koenkk/zigbee2mqtt/issues/25699

@AlexisPolegato you seem to be somehow connected to NodOn. Can you have a look?

AlexisPolegato, 2025-02-06:
These NodOn devices have 2 Firmwares to update ; it seems that Z2M does not handle that properly.
The issue seems to come from Z2M as it is working with other ecosystems.

Koenkk, 2025-02-06:
@AlexisPolegato that can be fixed with the `maxFileVersion`/`minFileVersion` (https://github.com/Koenkk/zigbee-OTA/blob/3658af07ec5d07806558977cca7d5d0b3553b81d/index.json#L7078)

Nardol, 2025-02-24:
Interesting, now the error is "No new image available".

Nardol, 2025-03-02:
Today I updated Zigbee2MQTT to version 2.1.3 then OTA was successful.
The update entity shows version 773 in Home Assistant, no error was logged by Zigbee2MQTT and no update is available now.