Koenkk/zigbee-OTA #680: Request for firmware update uue wall switch module

Dischi, 2025-02-08:
Hi,
your big hue  accessories update is missing the wall switch module (RDM001). The hue bridge updated the wall switch module from 1.0.5 to 2.77.35. Zigbee2mqtt does not show the update (it does not even show the device in the ota tab since the converter lacks ota=true).
Regards,
Dirk

woytekbode, 2025-03-26:
I also noticed that my Hue Wall Switch Modules on 1.05 are not visible on the OTA page.

rvanmaanen, 2025-05-25:
Yea same here, what I did was update to the latest version by using a Hue Hub I still have lying around then added them to Z2M. They are working fine, so not really a functional need for updating but I prefer to keep up-to-date with all my IoT devices for potential security fixes.