Koenkk/zigbee-OTA #691: SP24x Latest Firmware 1.9.28 Decimal Problem

epd5, 2025-02-11:
Hi there,

I'm assuming the issue is related to the patch notes  "CurrentSummationDelivered from 2 to 3 decimals: every Wh can now be tracked." - even after re-pairing the device, z2m & Home Assistant only show to 2 decimal places but seem to accept as if the 3rd was the 2nd, resulting in the value being multiplied by 10. As an example previous incremental device usage stats would be .02 w/H every 5 minutes the device was on, is now reporting as .11w/h every 5 minutes the same device is on. I think my understanding of the issue is right bit maybe I'm wrong, either way the CurrentSummationDelivered is wrong now.

Thanks,
epd5

Matt2k34, 2025-02-12:
Run OTA again, there's another firmware (1.9.30) which fixes that bug.

See here -> https://github.com/Koenkk/zigbee-herdsman-converters/issues/6747

epd5, 2025-02-12:
Thanks for this. I've upgraded a plug to 1.9.30, however the CurrentSummationDelivered still seems too high. Anyone else experiencing this? A 120W device has used over 1 Kilowatt in an hour...

Matt2k34, 2025-02-12:
Did you reconfigure the device after the FW upgrade? As it needs to now re-apply some settings.

(Yellow/orange) arrows on the main screen.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8715#issuecomment-2640979923

Skaars-fr, 2025-02-12:
I'm not on z2m but i'm facing the same issue (ZHA) 
should have think twice before updating my plugs

The energy consumption is far too hight, tried to update to 1.9.30, removed the device, add it back to HA, but it didn't help.

It was working perfectly before the update

epd5, 2025-02-12:
Re-paired the device, including the above re-interview step, the decimals are definitely off by a factor of 10 still. Luckily downgrading to 1.7 is possible.

Skaars-fr, 2025-02-12:
> Re-paired the device, including the above re-interview step, the decimals are definitely off by a factor of 10 still. Luckily downgrading to 1.7 is possible.

Is it possble to downgrade ?
May you please share the steps ?

epd5, 2025-02-12:
> > Re-paired the device, including the above re-interview step, the decimals are definitely off by a factor of 10 still. Luckily downgrading to 1.7 is possible.
> 
> Is it possble to downgrade ? May you please share the steps ?

I downgraded using the documentation in Zigbee2Mqtt. Not sure if you can do it in ZHA. Made the switch to Zigbee2Mqtt from ZHA a while ago, I haven't regretted it, would recommend.