Koenkk/zigbee-OTA #693: Request firmware update for Xiaomi Aqara E1 Wall Switch - lumi.switch.acn040

andremain, 2025-02-15:
I am having issues with my Xiaomi Aqara E1 Wall Switch - lumi.switch.acn040 being triggered by my IKEA buttons without any automations being setup. Could you please provided the latest firmware OTA for it so that i could check if the update fixes that issue?

References of the issue:

https://github.com/Koenkk/zigbee2mqtt/discussions/17244
https://github.com/Koenkk/zigbee2mqtt/issues/16918
https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072

Thank you.

Koenkk, 2025-02-16:
For someone who has both this device and the hub, see https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html in how to retrieve it

andremain, 2025-02-16:
> For someone who has both this device and the hub, see https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html in how to retrieve it

Hello and Thanks. I bought the E1 Hub but it will take around a month to arrive. I will try to extract the firmware as soon as i get it.

andremain, 2025-03-08:
Bad news. Just got the aqara e1 hub and added the aqara e1 wall switch to it. There is no newer firmware. I gust there is no fix for this..

closed 2026-04-08 by github-actions[bot]