Koenkk/zigbee-OTA #697: Run autodl more regularly

rohankapoorcom, 2025-02-18:
If a firmware file is uploaded by the manufacturer on a Monday, it seems weird to wait nearly a full week for the Sunday run to grab it. Can we run the process 1x / day instead?

Koenkk, 2025-02-19:
Im fine im with this, @Nerivec any objections?

Nerivec, 2025-02-19:
I'd rather keep the process weekly. It keeps the requests amount to various external servers lower, and in 99% of cases, it won't make any difference to run it more often anyway.
If an urgent update becomes available, we can trigger it manually, like we've already done in the past.

rohankapoorcom, 2025-02-19:
> I'd rather keep the process weekly. It keeps the requests amount to various external servers lower

Is this really an issue though? Before the updates were centralized here, each users Z2M reached out to the companies servers every time a check OTA update was hit. This is still a single centralized set of requests going to the external servers.

PR #708 (2025-03-01): Run auto download daily for specific manufacturers
As discussed with @Nerivec in the context of #697, a lot of manufacturers have extensive rate limiting on their API endpoints. For manufacturers that specifically request their OTAs being processed more frequently, we can run a custom job for this one.

Added a new action that will run daily and has an environment variable to populate the list of manufacturers. It intentionally does not trigger a release (keeping that for the weekly job)

Fixes: #697
closed 2025-03-01 by Nerivec