Koenkk/zigbee-OTA #734: Pvvx zigbee firmware no longer available?

hayvan96, 2025-04-02:
Hi, up to 0.1.2.3, OTA updated used to work for [LYWSD03MMC-z](https://www.zigbee2mqtt.io/devices/LYWSD03MMC-z.html#xiaomi-lywsd03mmc-z)

But new updates don't seem to be detected, I don't know if it's linked to the 1.4.X-->2.X migration. I have a sensor using 0.1.2.2, it should be updated to 0.1.2.3 or 0.1.2.4, but no update seems available. When I was still on Z2M 1.4.X, I was offered a 0.1.2.2.->0.1.2.3 OTA update for the same type of sensor.

thanks !

yannpub, 2025-04-03:
I just created a PR to upload firmwares v0.1.2.5 from yesterday. Just hoping I did it correctly.
https://github.com/Koenkk/zigbee-OTA/pull/735

yannpub, 2025-04-06:
Now that the PR has been merged, I did the OTA for both my LYWSD03MMC-z and my TS0201-z.

yannpub, 2025-04-07:
Be aware that following a change of SDK used to compile the firmware, the latest version requires a reset of the device and potentially a remove/re-add process.
https://github.com/pvvx/ZigbeeTLc/issues/179#issuecomment-2781007483

pvvx, 2025-04-07:
Version 0.1.2.5 has been removed due to bugs.
Ver 0.1.2.6. Fix - In some cases the timer did not turn off, which led to increased battery consumption. Changes in PollControl.

mrmaximas, 2025-04-07:
> Be aware that following a change of SDK used to compile the firmware, the latest version requires a reset of the device and potentially a remove/re-add process. [pvvx/ZigbeeTLc#179 (comment)](https://github.com/pvvx/ZigbeeTLc/issues/179#issuecomment-2781007483)

@yannpub please make new PR with the 1.2.6. Thanks!

mrmaximas, 2025-04-07:
@pvvx how to force the device to configure new measurement interval for exmple? I try it some times w/o result.

pvvx, 2025-04-07:
Do not put new firmware in the general list until other users have tested it! I do not have time to fully test more than 150 versions with different HW. It would take several months of reflashing, OTA and monitoring devices alone. For these reasons, testing of new versions is shifted to users.

mrmaximas, 2025-04-07:
@pvvx looks like I already see the problem, even with the 10 second interval set, the sensor doesn't show changes on the lcd for much longer, probably at least a minute, even when I breathe on it.

pvvx, 2025-04-07:
Version 0.1.2.5 has been removed because it does not work well with ZHA and Z2M.

yannpub, 2025-04-08:
I'll wait a bit for the various user feedbacks on 1.2.6 before doing a new PR.

mrmaximas, 2025-04-08:
> With two types of HW Xiaomi LYWSD03MMC B1.4 and B1.5 I checked.

is there any way to know the HW version with already flashed 1.2.6?

pvvx, 2025-04-08:
> [@pvvx](https://github.com/pvvx) is there any way to know the HW version with already flashed 1.2.6?

[image]

HW ID:
https://github.com/pvvx/pvvx.github.io/blob/master/README.md

mrmaximas, 2025-04-08:
@pvvx thanks! All of my devices is "hwVersion":10 = LYWSD03MMC-B1.5 with SHTV3 sensor, I wonder if the firmware will work if solder a SHT45-AD1B-R3 here ;)

mrmaximas, 2025-04-08:
@pvvx unable to force update from old 1.2.6 to latest 1.2.6. 
Update found, but it won't start.
```
[
    {
        "url": "/Users/max/zigbee2mqtt/data/1141-020a-01263001-Z03MMC.zigbee",
        "force": true
    }
]
```
[image]

pvvx, 2025-04-08:
On all my device variants with 0.1.2.5 OTA update to 0.1.2.6 was successful:

[image]

[image]

[image]

...

pvvx, 2025-04-08:
You may not encounter the changes in that update if you don't use the LYWSD03MMC contact closure often. Xiaomi MJWSD06MMC-z was added.

[image]

pvvx, 2025-04-08:
All changes are reflected in https://github.com/pvvx/ZigbeeTLc . 
I don't see any point in cluttering up the "issues" here anymore.

yannpub, 2025-04-11:
PR with Firmware 1.2.6 is complete.
As a reference, my LYWSD03MMC-z updated fine to 126 (from both 124 and 125), I had to delete/reinstall one TS0201-z.
I hope we can close this thread now ;-)

hayvan96, 2025-04-12:
> PR with Firmware 1.2.6 is complete. As a reference, my LYWSD03MMC-z updated fine to 126 (from both 124 and 125), I had to delete/reinstall one TS0201-z. I hope we can close this thread now ;-)

Absolutely, thanks for your PR! Updated to 0.1.2.6 from 0.1.2.5 all my devices.

closed 2025-04-12 by hayvan96

hayvan96, 2025-04-26:
Hi 0.1.2.7 seems out

yannpub, 2025-05-05:
I will not push the latest until there is an 'official' release in Victor's repo. If you want to test 0.1.2.7, what you can easily do is to add the following to your configuration yaml:

```
ota:
      zigbee_ota_override_index_location: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index_v0127.json
```