Koenkk/zigbee-OTA #736: Please add FW 1.0.10 for _TZ3000_x3ewpzyr for https://www.zigbee2mqtt.io/devices/XSH01B.html

mrmaximas, 2025-04-04:
FW 1.0.10 for _TZ3000_x3ewpzyr https://www.zigbee2mqtt.io/devices/XSH01B.html here:
{
            "autoSwitch": false,
            "canUpgrade": true,
            "controlType": 0,
            "currentVersion": "1.0.6",
            "desc": "1.0.10",
            "devType": 0,
            "diffOta": false,
            "fileSize": "315986",
            "firmwareDeployTime": 1730276298,
            "firmwareKey": "keyu7mte",
            "lastUpgradeTime": 0,
            "md5": "e6f014060c38957a537e5bfdcb55de6f",
            "timeout": 60,
            "type": 3,
            "typeDesc": "ZigBee Module",
            "upgradeStatus": 1,
            "upgradeType": 0,
            "upgradingDesc": "Please keep the power of the device connected during the upgrade process, please be patient.",
            "url": "https://euimagesd2h2yqnfpu4gl5.cdn5th.com/smart/firmware/upgrade/20240625/1719302262-oem_tl8258_zg_breaker_OTA_1.0.10.bin",
            "version": "1.0.10"
        }

[oem_tl8258_zg_breaker_OTA_1.0.10.bin.zip](https://github.com/user-attachments/files/19600476/oem_tl8258_zg_breaker_OTA_1.0.10.bin.zip)

Koenkk, 2025-04-04:
See https://github.com/Koenkk/zigbee-OTA?tab=readme-ov-file#adding-new-and-updating-existing-ota-files on how to add it

mrmaximas, 2025-04-04:
@Koenkk unfortunately I don't have my own laptop to do all this 🤦🏻‍♂️

mrmaximas, 2025-04-05:
@Koenkk done, I assume it will need to be corrected.

Koenkk, 2025-04-05:
Merged!

closed 2025-04-05 by Koenkk