Koenkk/zigbee-OTA #760: New Firmware Bosch Room-Thermostat II BTH-RM230Z available

h1nnak, 2025-04-25:
A new firmware is available, see here -> german: [click](https://www.bosch-smarthome.com/de/de/support/release-informationen/system-solutions/2025-04-02.html) + englisch: [click](https://www.bosch-smarthome.com/uk/en/release-notes/bosch-smart-home-system/2025-04-02.html)

Maybe someone can upload it? ;) @burmistrzak, @DerDreschner @Juan-ma or someone else?

There are a lot more firmware updates for other products.

DerDreschner, 2025-04-27:
Hey @h1nnak, as I don't own a BTH-RM230Z myself, I can't sniff the firmware unfortunately. If you're able to send one over, I can sniff the update and send it back to you.

But maybe @burmistrzak or @Juan-ma is able to sniff the update again.

Koenkk, 2025-07-10:
Added in https://github.com/Koenkk/zigbee-OTA/pull/805

closed 2025-07-10 by Koenkk

scroogie, 2025-07-10:
Just fyi, these were the three files I didn't add in #800 because they were already in the repo under a different name (as described in the PR).

Koenkk, 2025-07-10:
Somehow they were not in the `index.json` yet (which is needed to make them appear in z2m): https://github.com/Koenkk/zigbee-OTA/commit/2c7b60866e026960fe7a9b247a387ecfecd598f6

scroogie, 2025-07-20:
What to do with the duplicates, would you like a PR which deletes them?

Koenkk, 2025-07-20:
@scroogie which ones are duplicates? (yes please create a PR, then we can continue from there)

Koenkk, 2025-07-21:
@scroogie thanks, removed them!