Koenkk/zigbee-OTA #767: Bad INNR firmware

MSL-DA, 2025-05-09:
May I point out the latest firmware for INNR, version 1.9.34.

I would suggest that you remove this version, as there is an incredibly long transition from it turning on to reaching the desired brightness. This occurs even though the Transition is set to 0 in Z2M.

Another problem, and the worst one, is that the bulb turns on (perhaps 1%), then flashes once (100%), and then dims to the desired brightness in steps.

Affected devices are:
RB 247 T
RB 272 T

MSL-DA, 2025-05-09:
I just got a response from Innr. They can't recreate the behavior in ZHA. But it definitely happened between version 1.7.16 and version 1.9.34. This is the behavior I'm talking about.

https://github.com/user-attachments/assets/95bf1dd6-ea0a-4a94-bc69-86f56c7c69f9

MSL-DA, 2025-05-11:
This is related to in https://github.com/basnijholt/adaptive-lighting#readme home assistant.

MSL-DA, 2025-05-11:
I'm reopening it, as the behavior is also present if Transition is 1 or more. This applies to both Z2M and ZHA.

Koenkk, 2025-05-11:
CC: @gerryinnr

MSL-DA, 2025-05-12:
> Hi [@MSL-DA](https://github.com/MSL-DA) and [@Koenkk](https://github.com/Koenkk) thanks for bringing this to my attention. Let's see how we can fix this. If I understand correctly, this only happens if adaptive lighting is enabled, correct? Reading the info on the link provided above:

Thanks for the quick reply

It also happens without adaptive lighting, but only with Transition set to 1 or more in Z2M/ZHA

MSL-DA, 2025-05-12:
My English is not that good, but I will try to explain as best I can. Transition is the time from when the bulb is turned on until it reaches the desired brightness.

Transition is set in Z2M

[image]

MSL-DA, 2025-05-12:
This video is with Transition 1

https://github.com/user-attachments/assets/7a938ce5-9576-4d36-a9cd-8a76c4b65f35

MSL-DA, 2025-05-12:
It's a bit hard to reproduce as my iPhone does its best not to show the behavior when I'm recording :)

But I hope it's visible.

MSL-DA, 2025-05-12:
Here is a video with Transition 5, but where I turn off the light bulb. Just before it turns off, it flashes, then turns off.

https://github.com/user-attachments/assets/2f42df06-7d10-4c93-853e-55f5c1d47608

MSL-DA, 2025-05-13:
Okay... I've found out the behavior only occurs if Kelvin is set to 2,202K or 5,000K on the RB 247 T and 1,801K or 6,535K on the RB 272 T

MSL-DA, 2025-05-13:
@GerryInnr Thank you very much for the fantastic support here and by email.