Koenkk/zigbee-OTA #771: Multiple Issues with LYWSD03MMC-z

magic-happenz, 2025-05-15:
Hello,

I am using the Xiaomi LYWSD03MMC-z with custom firmware 3001-0125 and 3001-0126 and Zigbee2MQTT. Since I switched to the firmware, I have nothing but problems with several sensors. Some work well but some have issues such as:

- Missing bindings
- Occasionally stop reporting
- Battery dragged empty within days
- Failing OTA updates (sometimes crashing the Zigbee bridge

I tried addressing this problem already with Zigbee2MQTT (https://github.com/Koenkk/zigbee2mqtt/issues/26893) but I guess since its probably related to the firmware they won't really be able to help.

I hope someone here is able to help me as I am currently not able to really use the sensors given they are not reliable. I already tried downgrading the firmware but that didn't help much. Would the devices not even ignore manual OTA updates, I would have probably switched back to bluetooth firmware.

Any advice would be much appreciated.

yannpub, 2025-05-16:
You can try flashing them back to BLE, do a complete reset, then flash back to the ZigbeeTLC.
There has been some trouble with the OTA to 0125 since it was a new SDK, and it required some reset (but that was not working all the time).
I am running 0127 since some days without any issue.

The weird thing is your Zigbee bridge crashing. That would suggest you may have other issues in your setup.

magic-happenz, 2025-05-16:
> You can try flashing them back to BLE, do a complete reset, then flash back to the ZigbeeTLC.
> There has been some trouble with the OTA to 0125 since it was a new SDK, and it required some reset (but [...]

That's exactly what I was trying to do but the devices ignore my local OTA file and pull from Github no matter what. Any advise?

yannpub, 2025-05-16:
> magic-happenz>  left a comment > (Koenkk/zigbee-OTA#771) <https://github.com/Koenkk/zigbee-OTA/issues/771#issuecomment-2886323479>

Can you share your part of the config related to the OTA?

magic-happenz, 2025-05-16:
> Can you share your part of the config related to the OTA?
> 16 May 2025 at 12:27 by ***@***.***:
> […](#)

Sure. It should be alright as it worked once but unfortunately never again. I assume because of the versions.
In my zigbee2mqtt > `configuration.yaml` I added:
`
ota:
  zigbee_ota_override_index_location: my_index.json
`
Then in that file in the same folder I have:
`
[
    {
        "url": "ota/db15-0203-99993001-ATC_v46.zigbee",
        "force": true
    }
]
`
In the respective subfolder I have the file downloaded from https://github.com/devbis/z03mmc/blob/master/assets/db15-0203-99993001-ATC_v46.zigbee.
I couldn`t figure out the correct file stats so I decided to just put it on the server as according to the documentation the OTA is taking the information out of the file itself.

Last time it worked I could just search for updates and it installed the bluetooth firmware. Now it only finds the one in the GitHub repository.
I also tried sending a message with the device name to `https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#downgrade` via MQTT explorer as documented here: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#downgrade. But it then either could not find any available downgrade firmware or downgrades to a firmware also in the mentioned repository. Local config seems to be always ignored.

yannpub, 2025-05-16:
It is because the 203 file is for the devbis firmware. You should use 20a, then you'll be able to flash back to BLE, then back to ZigbeeTLc.

Image Type 	Device, note
0x0203 	LYWSD03MMC ver https://github.com/devbis/z03mmc
0x020A 	LYWSD03MMC ver https://github.com/pvvx/ZigbeeTLc

magic-happenz, 2025-05-16:
Can you send a direct link to the file? The ZigbeeTLc always links to https://github.com/pvvx/ATC_MiThermometer for 0x020A firmware so I am confused which is the right one now.

magic-happenz, 2025-05-16:
Are you saying I should revert to https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/db15-0203-99993001-ATC_v52.zigbee and then use https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html to install newest Zigbee firmware?

yannpub, 2025-05-16:
Yes, but you must use the correct file.
You are on ZigbeeTLc, thus you need the file 20a, and not 203.
Flash https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-020a-99993001-ATC_v52.zigbee to get back to BLE, then flash back to Zigbee using the telink flasher

magic-happenz, 2025-05-16:
Thanks man it really worked! Now the only problem still seems to be the bindings. They are partially not set automatically (missing power, temp and humidity) but unlike the previous versions the binding update works like a charm. Can you share you bindings and whether they got set correctly during the join?

magic-happenz, 2025-05-16:
Celebrated too early. I am still getting plenty of "failed to configure" .... `TABLE_FULL` messages.

yannpub, 2025-05-17:
I do not know how to get rid of this 'table full' issue. this is a recurrent topic, that seems related to z2mqtt, but I'm not qualified enough to tell you whether it is related to the ZigbeeTLc firmaware or to z2mqtt.

magic-happenz, 2025-05-17:
Yeah I saw some related issues and will have a look. But overall the sensors seem to run way more stable now and react to binding changes. Will close this issue for now. Again many thanks for the help.

closed 2025-05-17 by magic-happenz