Koenkk/zigbee-OTA #827: Gledopto GL-C-008P_V11376801 firmware issues

sosheskaz, 2025-08-05:
I downloaded and applied the new [GL-C-008P_V11376801](https://github.com/V4n1X/zigbee-OTA/blob/master/images/Gledopto/GL-C-008P_V11376801.ota) firmware that was recently added. The result is that the light strip:
* No longer responds appropriately to color temperature settings
* Misapplies color settings—seems like it's missing the red component maybe.

Koenkk, 2025-08-06:
I removed the firmware, could you try to [downgrade](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#downgrade-1) your device?

Koenkk, 2025-08-06:
Can you try again?

Crabang, 2025-08-06:
In my case its downgrading now, will give feedback as soon as the downgrade has finished.

salvq, 2025-08-06:
Downgrade completed but wrong colors and no cct etc.

Firmware info:
OK colors - firmware before upgrade: 11076802 20211020
NOK colors - firmware after upgrade: 11376801 20230413
NOK colors - firmware after downgrade V_1_7_A1 20210303

docBliny, 2025-08-07:
My downgrade on two units doesn't seem to have fixed things. I only have white/CCT on both and they aren't turning on at all anymore.

I have several others with the same model number and those have varying firmware versions:
* 10751203  - 20220824 (slightly newer than the affected units)
* 20551203 - 20240513 (several years newer)

malaakso, 2025-08-07:
Usually Gledopto's support (service@gledopto.com) provides correct OTA files when you explain the problem. You could directly ask for the version you had before the upgrade.

0rsa, 2025-08-07:
@malaakso By any chance, is it possible to retrieve the correct version number to send to geledopto before updating or am I cooked?

malaakso, 2025-08-07:
You would need to know the longer build ID, e.g. 11076802. That particular version is by the way available in one old pull request #124 , you can download it from there and use [local OTA index](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html) to push it.

AlienTekOriginal, 2025-08-07:
My 3 controllers are broken now too, if I don't have zigbee2mqtt and only ZHA can I fix this?

malaakso, 2025-08-07:
> My 3 controllers are broken now too, if I don't have zigbee2mqtt and only ZHA can I fix this?
> 
> I'm new to this, please be large with me. I'm not sure how to approach this..

Per [Zigpy docs](https://github.com/zigpy/zigpy/wiki/OTA-Configuration) you can specify local index for ZHA OTA as well to push above update manually. Best to ask in HA forums for more help.

AlienTekOriginal, 2025-08-07:
> Per [Zigpy docs](https://github.com/zigpy/zigpy/wiki/OTA-Configuration) you can specify local index for ZHA OTA as well to push above update manually. Best to ask in HA forums for more help.

I get no response on the discord/reddit forums, I see that i can use the index for ZHA but I have no idea what the next steps are to push the update manually...

AlienTekOriginal, 2025-08-07:
> Per [Zigpy docs](https://github.com/zigpy/zigpy/wiki/OTA-Configuration) you can specify local index for ZHA OTA as well to push above update manually. Best to ask in HA forums for more help.

The docs don't specify a way to "force" updates.

Chatgpt:

Because 21 < 22, ZHA will never push your GL-C-008P_V11376802_20230306.ota. Either:

Find a newer OTA (file_version > 22),

Flash by serial yourself, or

Patch the header (advanced, dangerous).

So us with ZHA are fucked?

V4n1X, 2025-08-08:
Hello,
I can clarify now a bit about the firmware situation.

GL-C-008P_V11376801 -> is for controller that have the sw_build_id ending with Vxxx76801 only

There is also another hardware layout that is ending with: Vxxx76802

You can fix your broken controllers with the firmware provided in my newest PR

Crabang, 2025-08-08:
> > Downgrade completed but wrong colors and no cct etc.
> > Firmware info:
> > OK colors - firmware before upgrade: 11076802 20211020
> > NOK colors - firmware after upgrade: 11376801 20230413
> > NOK colors - firmware after downgrade V_1_7_A1 20210303
> 
> This one helped me to correct the firmware and working as before [GL-C-008P_V11376802_20230306.zip](https://github.com/user-attachments/files/16676787/GL-C-008P_V11376802_20230306.zip)
> 
> Just follow the local update on [link](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files)
> 
> configuration.yaml
> 
> ```
> ota:
>     zigbee_ota_override_index_location: my_index.json
> ```
> 
> my_index.json
> 
> ```
> [
>     {
>         "url": "GL-C-008P_V11376802_20230306.ota",
>         "force": true
>     }
> ]
> ```

Thank for this, now my 6 devices are working as they should. Thank you all for your efforts, especially @Koenkk  for this project.

litecross91, 2025-09-29:
[GL-C-002P_V20951203_20241017_32013001.zip](https://github.com/user-attachments/files/22593516/GL-C-002P_V20951203_20241017_32013001.zip)
I got this, but it doesn't work for me due to a timeout error. This is the latest version. Try it and see if it works for you. For GL-C-002P version.