Koenkk/zigbee-OTA #838: Sonoff SNZB-02P is not showing up for OTA updates

Simonz82, 2025-08-19:
Hi, 
i have 5 Sonoff SNZB-02P is not showing up for OTA updates. 
my firmware is: 2.1.0, i know there is 2.2.0.
Could you please add an entry for my device revision?

other sonoff serie P there are in OTA update check.

thanks
Simone

Koenkk, 2025-08-19:
Can you try on the latest dev branch?

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Simonz82, 2025-08-22:
i have installed the edge version, now it sees in the list the SNZB-02P devices and sees that there is an update (as i said i have 0.0.00-2.1.00 and sees this: 0.0.00-2.2.00 ), when i click on update, it goes to error and does not update anything. there is however a bug i would say, the error is as follows:

z2m: Update of 'Temp Salone' failed (Device didn't respond to OTA request)
Update of 'Temp Salone' failed (Device didn't respond to OTA request)

Koenkk, 2025-08-24:
You probably need to wakeup the device right before clicking the update button.

Simonz82, 2025-08-24:
Surely this is the case, but how can I forcibly wake him up, so that he immediately clicks update?

Koenkk, 2025-08-24:
I'm not sure if the device has a button on it, but usually it wakes up when pressing that.

Simonz82, 2025-08-25:
ok, perfect, thanks to zigbee2mqtt edge i have updated all the thermometers.
in the OTA tab I also do not see the product SONOFF SNZB-05P, which however in the index I see listed, it is already updated to the latest version, but it is not listed for a possible check.
Other products that it does not find and in this case are NOT in the index file are:
TUYA: [TS0505B_1](https://www.zigbee2mqtt.io/devices/TS0505B_1.html#tuya-ts0505b_1)
MIBOXER: [FUT036Z](https://www.zigbee2mqtt.io/devices/FUT036Z.html#miboxer-fut036z)
SONOFF: [BASICZBR3](https://www.zigbee2mqtt.io/devices/BASICZBR3.html#sonoff-basiczbr3)
is it possible to integrate them ?

Thank you

MattiaMarchiorato, 2025-09-01:
@Koenkk are you planning to merge in main branch? 
I can't see OTA for 
SNZB-02P
SNZB-05P

Thanks

a1ex-ak, 2025-09-02:
[image]
This device determined automatically

[image]
here pressing the update firmware button does not help

What to do? Wake up?

MattiaMarchiorato, 2025-09-02:
@a1ex-ak you have to start the paring mode on the device. Then you will be able to lunch the update.

a1ex-ak, 2025-09-03:
> paring mode on the device

What kind of mode is this and how do I start it?

dkorunic, 2025-09-04:
I can confirm the above, SNZB-02P requires the following procedure:
- enable pairing process by holding button for 5 seconds
- red LED starts slowly flashing
- permit joining in ZB2MQTT
- immediately start OTA through ZB2MQTT

OTA itself takes around 30 minutes if the SNZB-02P is in reasonable proximity to ZB gateway.

closed 2025-12-30 by Koenkk

andrei-lazarov, 2026-01-07:
@pmalek I don't think they publish them, but they usually reply if you ask.