Koenkk/zigbee-OTA #863: Zigbee OTA update fail for Zigbee model: SPLZB-131

XeNoN6211, 2025-09-09:
Zigbee OTA update failed for Zigbee model: SPLZB-131 from 0.0.03-0.c.08 to 0.0.03-0.c.10. I waited two hours for something to happen, but it remained at 0%.

IDmedia, 2026-01-01:
@XeNoN6211 Did you find a solution to this? I have the same problem with a lot of, but not all of my smart plugs of the same model. They stay at 0% for hours while a few others are able to update. All of them are in the same room. I've also tried to run another Z2M instance just for updating them, but they still fail, so my best guess is that some devices are broken. They work, but are unable to update their software.

[image]

XeNoN6211, 2026-01-01:
Yes, I found the problem. Update Zigbee2MQTT to latest version.  This fixed my problem.

IDmedia, 2026-01-01:
Thanks for the reply. I updated to the version released today, so either that didn't fix or it it reintroduced the problem. I'll see if I can figure anything out. Same for edge. :)

IDmedia, 2026-01-01:
> maybe it helps to do them one by one and not all at once.

Nope, It's the same problem if I just update one at a time.

IDmedia, 2026-01-11:
@Koenkk 

1) I don't use the ota section at all. Not in the configuration.yaml nor home assistant addon config.
2) The config wasn't changed when updating from between the two versions (2.7.1 <-> 2.7.2)
3) I have created a fresh Z2M edge install (home assistant addon). The only thing I changed was to enable "availability" and OTA is broken, at least for that particular device. So the issue is present on the "default" configuration.

zigbee2mqtt/configuration.yaml:
```
version: 4
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: ...
  password: ...
  version: 4
  maximum_packet_size: 1048576
serial:
  rtscts: false
  port: tcp://192.168.1.101:6638
  adapter: ember
  baudrate: 115200
advanced:
  log_level: warning
  channel: 25
  network_key:
    - ...
  pan_id: 25509
  ext_pan_id:
    - ...
  log_file: log.log
frontend:
  enabled: true
  port: 8099
homeassistant:
  enabled: true
availability:
  enabled: true
devices:
  '0x7cc6b6fffe5e77db':
    friendly_name: Office Ceiling Light C
....
```

zigbee2mqtt_edge/configuration.yaml:
```
version: 4
mqtt:
  base_topic: zigbee2mqtt_edge
  server: mqtt://core-mosquitto:1883
  user: ...
  password: ...
serial:
  rtscts: false
  port: tcp://192.168.1.102:6638
  adapter: ember
  baudrate: 115200
advanced:
  log_level: info
  channel: 20
  network_key:
    - ...
  pan_id: 14694
  ext_pan_id:
    - ...
frontend:
  enabled: true
  port: 8099
homeassistant:
  enabled: true
availability:
  enabled: true
devices:
  '0x881a14fffefeafbd':
    friendly_name: Living Room Spots A
...

```