Koenkk/zigbee-OTA #871: [Bug] after adding new GL-C-008P Gledopto devices update checks always fail with JSON error

Ltek, 2025-09-14:
When I add a new device it never shows the current firmware, even if I force a reconfigure or reinterview. I have tried multiples today, all the same result.

[image]

the error I get in the log when clicking the button to check is this...

```
[2025-09-14 12:28:38] frontend:api: Sending {"topic":"bridge/request/options","payload":{"options":{"advanced":{"log_level":"error"}},"transaction":"hr1pv-1"}}
[2025-09-14 12:28:42] z2m: Failed to check if update available for '0xa4c13813048236ac' (Expected ',' or '}' after property value in JSON at position 75 (line 4 column 9))
```

Ltek, 2025-12-29:
Would love to see the Manual OTA improved
1. more robust/resilient -- UI based so these kinds of errors cannot occur
2. more informative -- what file is being displayed, and where from, and other usable info... nope, no clue.. I hope its the file in the OTA folder!  ;-)

Nerivec, 2025-12-29:
See https://github.com/Koenkk/zigbee-herdsman/issues/1608